Sensirion/embedded-uart-sps #10: Doubts regarding the mass concentration(ug/m3) of PM 10

niteshkadyan, 2018-11-21:
[image]

I ran the example provided for the linux host and got these values. I have doubts about the PM10 mass concentration reading. I don't understand why PM 10 value is smaller than PM 1, PM 2.5 and PM4? 

The data sheet defines PM10 to be all particles between 0.3 to 10 um in diameter. 

[image]

So, intuitively values of PM1 < PM2.5 < PM 4 < PM 10 right? Or am I missing something?

psachs, 2018-11-21:
@niteshkadyan yes you are right. The Mass concentration and number concentrations should always be larger the bigger the partical diameter gets as described in the datasheet.
The same holds for number concentrations.

We will have a look into this issue. Can you provide us some more information:
1. On what platform are you running the code? (e.g. raspberry pi, intel linux pc, ...)
2. Which driver release are you using?

niteshkadyan, 2018-11-21:
@psachs I tried multiple platforms: my Intel arch linux, Arduino mini and ESP32. On all the platforms I get the same result. On my linux host I ran the sample code provided in the github repo. I used sps30-uart based on the 0.9.1 tag in the master branch. 

And for Arduino and ESP32 I wrote my own UART based code, but got the exact same issue of PM10 being lower than others. 

Here is the screenshot of Arduino terminal based on my own code:

[image]

psachs, 2018-11-21:
@niteshkadyan Thank you for the extensive testing. I could not reproduce it with the sample on a intel linux machine.
Since you get similar results on multiple platforms I think the driver itself is working correctly.

I will check tomorrow with a couple of people if there are some known firmware issues regarding this. Where did you get your sample from?

niteshkadyan, 2018-11-21:
@psachs I bought it last week from [Arrow electronics](https://www.arrow.com/en/products/sps30/sensirion-ag):

niteshkadyan, 2018-11-21:
@psachs BTW, serial number for the sensor is AA42881345DD495D.

psachs, 2018-11-22:
@niteshkadyan Thank you for the information. We are currently tracking the sample and check the firmware and calibration information. 
Is it ok if we contact you directly on your gmail address?

Since it is not an issue with the driver itself I would like to close this issue on github

closed 2018-11-23 by psachs