Sensirion/embedded-uart-sps #13: "arduino_sps30_example.ino" does not compile

EKolev, 2018-12-12:
I installed embedded-uart-sps following the instructions in the README file.  I am using the arduino implantation from "sample-implementations/Arduino/sensirion_uart_implemenation.cpp" to compile the example arduino sketch and I am getting the following error:

"sensirion_uart_implementation.cpp:51:1: error: 'Uart' does not name a type
 Uart Serial2 (&sercom1, PIN_UART_RX, PIN_UART_TX, SERCOM_RX_PAD_0,
 ^ "

I am using an Arduino Uno board.

I tried the linux version on my Raspberry Pi and it works fine.

per1234, 2018-12-13:
That code is specific to the SAMD boards like the Zero, MKR, etc. AVR boards like your Uno don't have SERCOM.

EKolev, 2018-12-14:
I see, I deleted this line and used just 'Serial' in their implementation file and it worked. 

Thank you.

closed 2018-12-14 by EKolev

abrauchli, 2018-12-14:
@per1234, @EKolev thanks for the input, I'll add a comment the file

nit1995, 2019-01-09:
@EKolev  Can you share the code for Arduino implementation, please?