Sensirion/embedded-uart-sps #15: Problem with Arduino implementation

nit1995, 2019-01-09:
I am trying to implement the SPS30 on Arduino. I am using the sample code in embedded-uart-sps/sps30-uart/test_projects/Arduino/arduino_sps30_example/arduino_sps30_example.ino

I am getting conflicting declaration error as follows:

In file included from /home/dinesh/Arduino/libraries/embedded-uart-common/sensirion_uart.h:38:0,
                 from /home/dinesh/Arduino/libraries/embedded-uart-sps/sps30-uart/test_projects/Arduino/arduino_sps30_example/arduino_sps30_example.ino:33:
/home/dinesh/Arduino/libraries/embedded-uart-common/sensirion_arch_config.h:67:18: error: conflicting declaration 'typedef uint16_t u16'
 typedef uint16_t u16;
                  ^
In file included from /home/dinesh/arduino-1.8.7/hardware/arduino/avr/cores/arduino/Arduino.h:233:0,
                 from /home/dinesh/Arduino/libraries/embedded-uart-sps/sps30-uart/test_projects/Arduino/arduino_sps30_example/arduino_sps30_example.ino:31:
/home/dinesh/arduino-1.8.7/hardware/arduino/avr/cores/arduino/USBAPI.h:30:24: note: previous declaration as 'typedef short unsigned int u16'
 typedef unsigned short u16;
                        ^

Please help me in solving this error.

rnestler, 2019-01-09:
Hi @nit1995

Since your Arduino board library already seems to already provide these typedef's you can do as it states in the [comments above the typedef](https://github.com/Sensirion/embedded-uart-sps/blob/58e314e52e872452b56bb47b6750b95f04d0200e/embedded-uart-common/sensirion_arch_config.h#L58)
>     /**
>      * Comment out the following block if your platform already provides the
>      * type definitions u8, u16, u32, u64, s8, s16, s32, s64.
>      */

And remove the typedef for `u16`.

nit1995, 2019-01-10:
Thank you. I did that and the program is compiling without errors. To what pins of the Arduino must we connect the UART Rx and Tx? I am using the arduino_sps3_example.ino?

rnestler, 2019-01-10:
That depends on what Arduino board your are using. See https://www.arduino.cc/reference/en/language/functions/communication/serial/.

yukimach, 2019-11-27:
Hi,
I'm getting the: 
conflicting declaration 'typedef uint16_t u16'

when i comment this out
// typedef uint16_t u16

I get
unknown type name 'u16'

Where am I going wrong?

rnestler, 2020-01-22:
@yukimach This shouldn't happen with the newer releases (>2.0.0) since we changed to using the `uint16_t` types directly. Please open a new issue if you still have any troubles.