Sensirion/embedded-uart-sps #16: Problem with implementation in Feather 32u4

nit1995, 2019-01-09:
I removed the following lines from sensirion_uart_implementation.cpp as suggested in one of the issues.
Uart Serial2 (&sercom1, PIN_UART_RX, PIN_UART_TX, SERCOM_RX_PAD_0,
              UART_TX_PAD_2);

And I changed Serial2 to Serial. 
I am now getting the following error:
/home/dinesh/Arduino/sketch_jan09c/sketch_jan09c.ino: In function 's16 sensirion_uart_open()':
sketch_jan09c:71:32: error: 'PIO_SERCOM' was not declared in this scope
     pinPeripheral(PIN_UART_TX, PIO_SERCOM);
                                ^
sketch_jan09c:71:42: error: 'pinPeripheral' was not declared in this scope
     pinPeripheral(PIN_UART_TX, PIO_SERCOM);
                                          ^
exit status 1
'PIO_SERCOM' was not declared in this scope

I am using a Feather 32u4 Arduino based board. Please help me solve these errors.

rnestler, 2019-01-10:
According to https://learn.adafruit.com/adafruit-feather-32u4-basic-proto?view=all#logic-pins-2-9 it looks like you can use pin 0 and 1 with Serial1.

Then just use `Serial1.begin()` and remove the `pinPeripheral` calls.

abrauchli, 2019-01-11:
Arduino just compiles all files it finds which is a bit suboptimal in this case. You'll have to remove `sps30_example_usage.c`, I'm assuming you're using some kind of `.ino` file. You can use the example as a base for what functions to call on the generic sps driver..

I don't know which is the correct serial interface to use for which pins on your board since this is very platform specific. For general Arduino programming support you'll be much better off asking in an Arduino forum. But once you get it working, we'd be happy to include your findings on how to get the example working for your board. I'll leave the issue open for a while.

nit1995, 2019-01-11:
Thanks, I kinda figured that out. I removed the sps30_example_usage.c file. I then added 
#define NULL 0 in sps30.c . 
The program is now compiling without errors. There is a problem with interfacing it to the hardware. I will update any progress.

abrauchli, 2019-02-01:
@nit1995 did you get it working? If so, would you provide us with a sample or post a pull request?

rnestler, 2019-04-03:
@nit1995 Do you have any updates?

nit1995, 2019-04-08:
@rnestler Nope. I stopped working on the project. Instead of Feater, we integrated it with another microcontroller.

rnestler, 2019-05-15:
@nit1995 I'll close this for now. Feel free to re-open or open a new issue if you encounter any problems :slightly_smiling_face:

closed 2019-05-15 by rnestler