Sensirion/embedded-uart-sps #18: Node MCU Implementation

Alex507507507, 2019-01-10:
Hi
I am trying to read out the values of the sps30 with an NodeMCU and the arduino IDE.

First I could not integrate the library as usual with the Sketch Zip Integrate function.
In the end I copied the files sps30-uart and embedded-uart-common into the arduino library folder and included them.

Now I am using the sps30_example.
I changed the sensirion_uart_implementation file adding the functions of the sample sensirion_uart_implementation.cpp: Here I did Not add 
Uart Serial2 (&sercom1, PIN_UART_RX, PIN_UART_TX, SERCOM_RX_PAD_0,
              UART_TX_PAD_2);

void SERCOM1_Handler() {
  Serial2.IrqHandler();
}
because there are errors and I guess its not needed.

I changed the c file into an cpp file, otherwise the serial functions did not work and I removed some serial 2 to serial 1:

Here is the sensirion uart implementation version
/////////////////////////////////////////////////////////////////////////////////////////////

// needed for delay() routine
#include <Arduino.h>
#include "wiring_private.h" // pinPeripheral() function

#ifdef __cplusplus
extern "C" {
#endif

#include "sensirion_arch_config.h"
#include "sensirion_uart.h"

#define BAUDRATE 115200  // baud rate of SPS30
#define PIN_UART_RX 11
#define PIN_UART_TX 10
/*
 * INSTRUCTIONS
 * ============
 *
 * Implement all functions where they are marked with TODO: implement
 * Follow the function specification in the comments.
 */
 
 

 
 
/**
 * sensirion_uart_open() - initialize UART
 *
 * Return:      0 on success, an error code otherwise
 */
s16 sensirion_uart_open() {

    while (!Serial) {
        delay(100);
    }
    return 0;
}

/**
 * sensirion_uart_close() - release UART resources
 *
 * Return:      0 on success, an error code otherwise
 */
s16 sensirion_uart_close() {
    // TODO: implement x
	    Serial.end();
	
    return 0;
}

/**
 * sensirion_uart_tx() - transmit data over UART
 *
 * @data_len:   number of bytes to send
 * @data:       data to send
 * Return:      Number of bytes sent or a negative error code
 */
s16 sensirion_uart_tx(u16 data_len, const u8 *data) {
    // TODO: implement x
	    return Serial.write(data, data_len);
    return 0;
}

/**
 * sensirion_uart_rx() - receive data over UART
 *
 * @data_len:   max number of bytes to receive
 * @data:       Memory where received data is stored
 * Return:      Number of bytes received or a negative error code
 */
s16 sensirion_uart_rx(u16 max_data_len, u8 *data) {
    // TODO: implement
    s16 i = 0;

    while (Serial.available() > 0 && i < max_data_len) {
        data[i] = (u8)Serial.read();
        i++;
    }

    return i;
}

/**
 * Sleep for a given number of microseconds. The function should delay the
 * execution for at least the given time, but may also sleep longer.
 *
 * Despite the unit, a <10 millisecond precision is sufficient.
 *
 * @param useconds the sleep time in microseconds
 */
void sensirion_sleep_usec(u32 useconds) {
    // TODO: implement x
	    delay((useconds / 1000) + 1);
}

}
////////////////////////////////////////////////////////////////////////////////////////////
The Serial Monitor shows this message:
~ ⸮+~ probe failed
~ ⸮+~ probe failed
~ ⸮+~ probe failed

I connected the rx/tx to the rx/tx of the nodemcu and the vin to the power supply and ground to ground as written in the data sheet.

The Ventilator is not turning.

Do you have any recommendations?

Thank you very much in advance and best regards
Sebastian

rnestler, 2019-01-11:
Hi @Alex507507507 I edited your comment such that the code has syntax highlighting, I hope you don't mind.

> The Serial Monitor shows this message:
> ```
>~ ⸮+~ probe failed
>~ ⸮+~ probe failed
>~ ⸮+~ probe failed
>```

It looks like you are using the same serial port for the connection to the sps30 as is used for the USB serial monitor. This won't work since there will be conflicts.
Is there another Serial port available on your Arduino board? See also https://www.arduino.cc/reference/en/language/functions/communication/serial/ for Serial ports available for common Arduino boards.

> s16 sensirion_uart_open() {

You should have a `Serial1.begin(BAUDRATE)` there, where `Serial1` may need to change depending on your Arduino board.

If there is no additional Serial port available you may try to use the https://www.arduino.cc/en/Reference/SoftwareSerial library.

Alex507507507, 2019-01-11:
Hello 
@rnestler , thank you for your answer and your advise to use the software serial and to highlight the code.

I added the software serial 2 to the sensirion_uart_implementation to read the values. D1 and D2 are the Pins of the NodeMCU. The implementations are the implementations from the sample. 

```ruby
require 'redcarpet'
markdown = Redcarpet.new("Hello World!")
puts markdown.to_html

/*
 * Copyright (c) 2018, Sensirion AG
 * All rights reserved.
 *
 * Redistribution and use in source and binary forms, with or without
 * modification, are permitted provided that the following conditions are met:
 *
 * * Redistributions of source code must retain the above copyright notice, this
 *   list of conditions and the following disclaimer.
 *
 * * Redistributions in binary form must reproduce the above copyright notice,
 *   this list of conditions and the following disclaimer in the documentation
 *   and/or other materials provided with the distribution.
 *
 * * Neither the name of Sensirion AG nor the names of its
 *   contributors may be used to endorse or promote products derived from
 *   this software without specific prior written permission.
 *
 * THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS "AS IS"
 * AND ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT LIMITED TO, THE
 * IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE ARE
 * DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT HOLDER OR CONTRIBUTORS BE LIABLE
 * FOR ANY DIRECT, INDIRECT, INCIDENTAL, SPECIAL, EXEMPLARY, OR CONSEQUENTIAL
 * DAMAGES (INCLUDING, BUT NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR
 * SERVICES; LOSS OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER
 * CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, STRICT LIABILITY,
 * OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE) ARISING IN ANY WAY OUT OF THE USE
 * OF THIS SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.
 */

#include "sensirion_arch_config.h"
#include "sensirion_uart.h"

#include <SoftwareSerial.h>

#include <Arduino.h>

SoftwareSerial Serial11(D1, D2);

/*
 * INSTRUCTIONS
 * ============
 *
 * Implement all functions where they are marked with TODO: implement
 * Follow the function specification in the comments.
 */

/**
 * sensirion_uart_open() - initialize UART
 *
 * Return:      0 on success, an error code otherwise
 */
s16 sensirion_uart_open() {
	Serial11.begin(9600);
	
	
    while (!Serial11) {
        delay(100);
    }
    return 0;
}

/**
 * sensirion_uart_close() - release UART resources
 *
 * Return:      0 on success, an error code otherwise
 */
s16 sensirion_uart_close() {
    Serial11.end();	
    return 0;
}

/**
 * sensirion_uart_tx() - transmit data over UART
 *
 * @data_len:   number of bytes to send
 * @data:       data to send
 * Return:      Number of bytes sent or a negative error code
 */
s16 sensirion_uart_tx(u16 data_len, const u8 *data) {
    return Serial11.write(data, data_len);

}

/**
 * sensirion_uart_rx() - receive data over UART
 *
 * @data_len:   max number of bytes to receive
 * @data:       Memory where received data is stored
 * Return:      Number of bytes received or a negative error code
 */
s16 sensirion_uart_rx(u16 max_data_len, u8 *data) {
	    s16 i = 0;
	
    while (Serial11.available() > 0 && i < max_data_len) {
        data[i] = (u8)Serial11.read();
        i++;
    }

    return i;
}

/**
 * Sleep for a given number of microseconds. The function should delay the
 * execution for at least the given time, but may also sleep longer.
 *
 * Despite the unit, a <10 millisecond precision is sufficient.
 *
 * @param useconds the sleep time in microseconds
 */
void sensirion_sleep_usec(u32 useconds) {
    delay((useconds / 1000) + 1);	
}
```

I tried adding the start function to the arduino_sps30_example, but I could not find a way to turn on the ventilator and to start communicating. I also tried to add replace some "serial" to "Serial11" in the sps30 library. 
How to indicate the sensor to start measuring properly?

Thank you in advance and best regards
Alex

rnestler, 2019-01-12:
> `Serial11.begin(9600);`

This looks like the wrong baudrate. Try to use the [defined `BAUDRATE`](https://github.com/Sensirion/embedded-uart-sps/blob/58e314e52e872452b56bb47b6750b95f04d0200e/embedded-uart-common/sample-implementations/arduino/sensirion_uart_implementation.cpp#L42) value which is 115200.

Alex507507507, 2019-01-12:
Hello, 
@rnestler Thank you for your answer and your recommendations. 

I changed the bautrate to 115000.
- The ```sensirion_uart_open(); ``` function is inside the ```setup``` function
- The ```serial monitor``` shows the error message 
``` probe failed, probe failed ...``` 

Therefore the sps_probe has an error.

if I remove that while loop the ``` sps30_start_measurement``` 
also returns the error message. 
``` read measurement failed, read measurement failed ...``` 

when I print out the value before
``` sps30_read_measurement(&measurement);``` then ret=0 
when printing it out after 
``` sps30_read_measurement(&measurement);``` then ret=-2.

What about the fourth pin for interface select: it says it should be "left floating" for uart.
From googling I thought it should be just hanging without connecting it. Is it right?
The ventilator is not turning.

(Somehow I could make the ventilator turn somehow before, however, I didn't realize it so I don't know how I made it run before... couldn't repeat that.. At least it's not broken.)

Any other ideas?

Thank you a lot in advance and best regards
Alex

rnestler, 2019-01-14:
> What about the fourth pin for interface select: it says it should be "left floating" for uart.
From googling I thought it should be just hanging without connecting it. Is it right?

Yes, "left floating" means to not connect it.

>  The ventilator is not turning.

The ventilator will only start when the device is measuring, to save power if not measuring. So as long as `sps30_start_measurement` fails it won't.

Sebastiang507, 2019-02-08:
Hello 
please find the working example. Thank you for your help, sorry for answering that late. There is still a problem with sensirion_uart_close() when transfering the data to a server, however without it it works. 

```ruby

#include <sensirion_arch_config.h>
#include <sensirion_shdlc.h>
#include <sensirion_uart.h>

/*
   Copyright (c) 2018, Sensirion AG
   All rights reserved.

   Redistribution and use in source and binary forms, with or without
   modification, are permitted provided that the following conditions are met:

 * * Redistributions of source code must retain the above copyright notice, this
     list of conditions and the following disclaimer.

 * * Redistributions in binary form must reproduce the above copyright notice,
     this list of conditions and the following disclaimer in the documentation
     and/or other materials provided with the distribution.

 * * Neither the name of Sensirion AG nor the names of its
     contributors may be used to endorse or promote products derived from
     this software without specific prior written permission.

   THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS "AS IS"
   AND ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT LIMITED TO, THE
   IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE ARE
   DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT HOLDER OR CONTRIBUTORS BE LIABLE
   FOR ANY DIRECT, INDIRECT, INCIDENTAL, SPECIAL, EXEMPLARY, OR CONSEQUENTIAL
   DAMAGES (INCLUDING, BUT NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR
   SERVICES; LOSS OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER
   CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, STRICT LIABILITY,
   OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE) ARISING IN ANY WAY OUT OF THE USE
   OF THIS SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.
*/
#include <Arduino.h>

#include "sensirion_uart.h"
#include "sps30.h"

void setup() {
    Serial.begin(115200);

    
    while (!Serial) {
      delay(100);
    }

    // use built-in LED to show errors
    sensirion_uart_open();
    s16 sps30_start_measurement();
}

void loop() {
    struct sps30_measurement measurement;
    s16 ret;

Serial.println(ret);

    while (sps30_probe() != 0) {
        Serial.write("probe failed\n");
        delay(1000);
    }

    /* start measurement and wait for 10s to ensure the sensor has a
     * stable flow and possible remaining particles are cleaned out */
    
[... 149 lines cut ...]
    while (Serial11.available() > 0 && i < max_data_len) {
        data[i] = (u8)Serial11.read();
        i++;
    }

    return i;
}

/**
 * Sleep for a given number of microseconds. The function should delay the
 * execution for at least the given time, but may also sleep longer.
 *
 * Despite the unit, a <10 millisecond precision is sufficient.
 *
 * @param useconds the sleep time in microseconds
 */
void sensirion_sleep_usec(u32 useconds) {

    delay((useconds / 1000) + 1);	
}

```

Best regards
Alexander

rnestler, 2020-07-02:
I'll close this, since there wasn't any activity and the solution from @Sebastiang507 seems to work.

closed 2020-07-02 by rnestler