Sensirion/embedded-uart-sps #19: config for Arduino

paultanner, 2019-01-15:
I have followed the instructions (on MacOS) and got a viable folder in my Arduino setup - almost :-(
On building this I get typedef conflicts, possibly  because I have not adjusted sensirion_arch_config.h

Errors such as "previous declaration as 'typedef short unsigned int u16' typedef unsigned short u16"

If I delete the u16 typedef I get "sensirion_uart.h:61: error: unknown type name 'u16'"

Same issue on Leonardo, Mega and Uno.

Any suggestions pls?

rnestler, 2019-01-15:
> Errors such as "previous declaration as 'typedef short unsigned int u16' typedef unsigned short u16"
> If I delete the u16 typedef I get "sensirion_uart.h:61: error: unknown type name 'u16'"

Could you try and replace the typedef in `sensirion_arch_config.h` with the previous declaration of the Arduino library: `typedef short unsigned int u16`.

paultanner, 2019-01-15:
Thx @rnestler 
That did the trick
```
//typedef uint16_t u16;
typedef short unsigned int u16;
```
Compiles for 3 arduinos that I tried.

psachs, 2019-01-15:
The sample implementation was done for a Arduino M0 Pro with a ATSAMD21 and will not work on any Arduino board. Depending on the Arduino board a different approach is needed.

Since each serial port can only be used for one device, we can not use the same serial port for the SPS30 as we use to connect the PC with the Arduino for programming and debugging. 

Since Arduino has different variations there is no generic solution that works for all boards. If your board already has a second (or even more) serial ports, you can just use that one.

You can check if your boards support the SoftwareSerial or AltSoftSerial library
https://www.arduino.cc/en/Reference/SoftwareSerial
https://www.pjrc.com/teensy/td_libs_AltSoftSerial.html

Let us know what Arduino boards you are using and if one of those libraries worked. Of course you have to adapt the sample implementation to use the library.

paultanner, 2019-01-15:
Thx @psachs 
The boards I have do support softserial but I was thinking of using a leonardo which has 2 hardware serial ports.
So, if I understand you, no one has contributed implementations that use these standard boards  so I need to write the code into sensirion_uart_implementation.c
Correct?

abrauchli, 2019-01-15:
> Thx @psachs
> The boards I have do support softserial but I was thinking of using a leonardo which has 2 hardware serial ports.
> So, if I understand you, no one has contributed implementations that use these standard boards so I need to write the code into sensirion_uart_implementation.c
> Correct?

Spot on. 

For the Leonardo it actually shouldn't be that bad:

Quote from https://www.arduino.cc/reference/en/language/functions/communication/serial/
> The Arduino Leonardo board uses Serial1 to communicate via TTL (5V) serial on pins 0 (RX) and 1 (TX). Serial is reserved for USB CDC communication. For more information, refer to the Leonardo getting started page and hardware page.

Most of the example may still be valid if you remove the definition for `Serial2` on L56, remove the `SERCOM1_Handler` function on L59 and replace all remaining instances of `Serial2` with `Serial1`. One notable exception is `sensirion_uart_open` which should become much simpler:

```cpp
s16 sensirion_uart_open() {
  Serial1.begin(BAUDRATE);
  return 0;
}
```

The defines `PIN_UART_RX` and `PIN_UART_TX` are wrong (should be 0 and 1) but they're also not used anymore..

Good luck, let us know if that works.

abrauchli, 2019-01-16:
Hi @paultanner, thanks for the feedback. We are trying to improve the situation for Arduino and provide ready-to-use samples for different boards, so we'd be glad to get a tested config for yours as well. Feel free to take a look or review the Arduino-specific README https://github.com/Sensirion/embedded-common/pull/15 (it's for the i2c variants for now, but most things apply to the UART version as well).

abrauchli, 2019-02-01:
Related to #17

psachs, 2022-01-11:
We do have Arduino libraries for most of our newer sensors available.

closed 2022-01-11 by psachs