Sensirion/embedded-uart-sps #20: Sample for STM32 (HAL)

olirehacek, 2019-01-17:
Hello,
For past few days, I am struggling with implementation on our STM32L0xx-based board. Can you please post some sample how to process data from the sensor using HAL libraries from ST ?

Thank you.

olirehacek, 2019-01-21:
> Hi @olirehacek
> 
> Which HAL libraries do you mean? Do you have a link to a description from ST?

Hi @rnestler , appreciate the fast reply.

I mean standard STMicroelectronics HAL Libraries that are supplied by the vendor together with CubeMX  Code Initialization Software. I have a Nucelo Development Board and just trying to slap together simple air quality measuring device into our office.

I have managed to send all the commands to the sensors and it works perfectly, but receiving the data is a problem. My receiving buffer is always filled with zeros.
I am totally new to the embedded developing so in case I am doing something completely stupid I'm sorry :D

`
uint8_t buf_read[6] = { 0x7e, 0x00, 0x03, 0x00, 0xfc, 0x7e };
	HAL_UART_Transmit(&huart2, buf_read, 8, 1000);
	//Wait for start byte to be received (0x7E)
   for(int i = 0 ; i < 100 ; i++)
	{		
		HAL_UART_Receive(&huart2, (uint8_t *)RxBuffer, 1, 1000);
		if (RxBuffer[0] == 0x7E)
			break;
	}
	//Get rest of packet
	HAL_UART_Receive_DMA(&huart1, RxBuffer, 50);
`

olirehacek, 2019-02-03:
> @olirehacek I'm curious if you got it working and if so, would you provide us with a working sample (pull-request welcome)?

Hi. I am out of the office until Tuesday. Hopefully, I can get it running by the end of the week - i will post some results by then.

rnestler, 2019-04-03:
@olirehacek Did you find some time to test again?

Shubhanshu2student, 2019-06-04:
Hello there,
I have got this sensor in the past few days, and I'm trying to get it implemented with STM32F030R8 discovery board using USART.
Firstly I've tried to get the Serial ID with the given SHDLC driver, transmission buffer is correct while receiving sensor is not responding with any data byte.
Any help is appreciated.

rnestler, 2019-06-04:
>  Firstly I've tried to get the Serial ID with the given SHDLC driver, transmission buffer is correct while receiving sensor is not responding with any data byte.

Can you verify with a logic analyzer or oscilloscope what is happening on the bus?

>  Below is my rx and tx functions.

The implementation doesn't look quit correct, see my comments:
```c
    uint8_t tx_data = *tx_buff++;
    USART2->TDR = tx_data;
    // Here you should probably wait for the date to get transmitted, or does USART_ClearITPendingBit do that?
    USART_ClearITPendingBit(USART2,USART_ISR_TC);
```

The `IRQHandler` will just override `rx_buf[0]` with every byte received.
```
void USART2_IRQHandler(void)
{
  uint8_t ind=0; // index is always 0 when the IRQ gets called
  USART_ClearITPendingBit(USART2, USART_IT_ORE);
  if((USART_GetITStatus(USART2,USART_IT_RXNE))==SET )
  {    
    Temp = ((USART_ReceiveData(USART2))&0xff);
    rx_buff[ind++] = Temp;
    if(ind >= (sizeof(rx_buff)-1))
      ind=0;
    USART_ClearITPendingBit(USART2,USART_IT_RXNE);
  }
}
```

The `sensirion_uart_rx` function just copies to a local buffer and not to the target buffer. Also you can't now how many bytes there are in the `rx_buff`.
```
uint8_t rx_buff[100];
uint16_t sensirion_uart_rx(uint16_t max_data_len, uint8_t *data)
{
  uint8_t data_buff[100];
  if(max_data_len == 0)
    return -1;
  memcpy(data_buff,rx_buff,max_data_len); // this just copies to the local buffer
  data = data_buff; // this doesn't write to the target buffer, just reassign the pointer locally
  return sizeof(data_buff); // this will always return 100
}
```

Isn't there an UART library available for your platform? This should handle all the low level stuff like interrupt handling for you. Maybe try using the "STMicroelectronics HAL Libraries" like @olirehacek did.

Shubhanshu2student, 2019-06-05:
thanks for your reply, surely I'll check things out and update...
About libraries, I'm using STM Std Peripheral Libraries, to learn more in deep about the peripheral programming.

Shubhanshu2student, 2019-06-07:
hello there, 
I got the sensor working giving me the measured values, Now I'm having some testing issues,
1. How we can verify the values are correct?
2. On testing with Air purifier and Sensor in a closed cardboard box the sensor was giving me some constant values like 9-6 for all the parameters.

Shubhanshu2student, 2019-06-10:
The given link is having a video in which I'm getting the data from the sensor. Are these values correct?
If not, suggest the improvement I need to do.
https://drive.google.com/open?id=1_k3cfPWxw5sNjZvufEVvvn36u5UUYHJD

sisanche, 2021-02-01:
@Shubhanshu2student while opening the google drive link showing an error.
Could you please post the link or is it possible to share the source files?