Sensirion/embedded-uart-sps #21: SPS30 does not go return to idle when calling sps30_stop_measurement() in sps30.c

utilebit, 2019-01-31:
First, thank you for providing this driver and implementation for the SPS30.

When the sps30_stop_measurement method in the sps30.c is called the SPS30 sensor's fan still runs and does not return to idle.
```
s16 sps30_stop_measurement() {
    u8 param_buf[] = SPS_SUBCMD_MEASUREMENT_STOP;

    return sensirion_shdlc_tx(SPS_ADDR, SPS_CMD_MEASUREMENT,
                              sizeof(param_buf), param_buf);
}
```

The sps30_stop_measurement method shown above sends a length of 1 and the stop command 0x00 as a sub-command to the sensirion_shdlc_tx method. The SPS30 datasheet for the stop command provides the following example frame for the stop command:
```
0x7E 0x00 0x01 0x00 0xFE 0x7E
```
This example frame uses the command 0x01 instead of 0x00 and since it does not have sub-commands it uses a length of zero. The current method does not send the example frame for MOSI described in the datasheet. For this reason, I believe this is a bug.

I was able to fix the bug by adding the constant:
```
#define SPS_CMD_STOP_MEASUREMENT 0x01
```

and modifying the method to be:
```
s16 sps30_stop_measurement() {
    return sensirion_shdlc_tx(SPS_ADDR, SPS_CMD_STOP_MEASUREMENT, 0, NULL);
}
```

abrauchli, 2019-02-01:
Hi @utilebit, many thanks for the detailed report and the proposed fix. Would you like to submit a pull request or are you happy with leaving it to us?

When submitting a patch, consider that `SPS_CMD_MEASUREMENT` would also better be renamed to `SPS_CMD_START_MEASUREMENT`.

utilebit, 2019-02-07:
You are welcome. I submitted a pull request with the fix and the renamed constant.

abrauchli, 2019-02-07:
Thanks, merged as 6e3256e8d4e63dcc49a6e5f322467fabfb8a2400 (#23)

closed 2019-02-07 by abrauchli