Sensirion/embedded-uart-sps #24: Adafruit Feather 32u4 Bluefruit LE: Unable to receive data

hijoumake, 2019-02-15:
Using the provided sample implementation `sensirion_uart_implementation.cpp` modified to use the hardware serial port `Serial1` available on the Feather 32u4 platform and using `typedef short unsigned int u16;` in `sensirion_arch_config.h` , I've been able to get the sample code to compile, but unable to get `sps30_probe()` to return successfully. 

The return value of `sps30_probe()` indicates a failure corresponding to the `SENSIRION_SHDLC_ERR_MISSING_STOP` error code. I've further isolated the origin of the error code to this section of `sensirion_shdlc.c` https://github.com/Sensirion/embedded-uart-sps/blob/6e3256e8d4e63dcc49a6e5f322467fabfb8a2400/embedded-uart-common/sensirion_shdlc.c#L178

I've also tried removing the RX and TX pins one at a time on the hardware side which resulted in the return value of `sps30_probe()` changing to a value corresponding to `SENSIRION_SHDLC_ERR_MISSING_START` in both cases (TX pin floating and RX pin floating, respectively). 

Further, I've tested the SPS30 unit using a USB sensor bridge and the ControlCenter application. It appears to be functioning normally in this context.

I'd appreciate suggestions for further troubleshooting steps.

rnestler, 2019-02-20:
Hi @hijoumake

> I had to remove the call to `sps30_probe()` to make the sample work, but all the other commands seemed to execute correctly, including `sps30_get_serial()` which is called by `sps30_probe`. I'm unsure why `sps30_probe()` was returning a non-zero value.

Maybe it is a timing issue, that `sps30_get_serial` doesn't work immediately after start-up. Can you try delaying a few seconds before doing the probe?

> The return value of sps30_probe() indicates a failure corresponding to the SENSIRION_SHDLC_ERR_MISSING_STOP error code. I've further isolated the origin of the error code to this section of sensirion_shdlc.c

Hmm `SENSIRION_SHDLC_ERR_MISSING_STOP` gets returned at multiple places:
https://github.com/Sensirion/embedded-uart-sps/blob/6e3256e8d4e63dcc49a6e5f322467fabfb8a2400/embedded-uart-common/sensirion_shdlc.c#L180
https://github.com/Sensirion/embedded-uart-sps/blob/6e3256e8d4e63dcc49a6e5f322467fabfb8a2400/embedded-uart-common/sensirion_shdlc.c#L190
https://github.com/Sensirion/embedded-uart-sps/blob/6e3256e8d4e63dcc49a6e5f322467fabfb8a2400/embedded-uart-common/sensirion_shdlc.c#L197

Are you sure it happened in that exact location? We should probably add more error codes for these error conditions, because only the one at L197 seems to match the error condition condition.

hijoumake, 2019-02-22:
I added a 5 second delay and I am still receiving the same results. I'm sure that it is failing at L197 because I replaced the return value with a custom error value to isolate which condition was causing the function to return.

The input/output I'm receiving from the Arduino serial console from my debugging statements is as follows:
`
Bytes written: 7E 0 D0 1 3 2B 7E 
Bytes sent: 7
Bytes received: 7E D0 7D 4C BA 43 A6 B2 42 9A 30 4C 13 15 31 2 A6 FF 
Bytes read: 18
`
The MOSI frame seems to be correct, but the MISO frame does not make sense.

rnestler, 2019-02-22:
> I'm sure that it is failing at L197

In the previous comment you said you tracked down the error to "Line 178":

> I've further isolated the origin of the error code to this section of `sensirion_shdlc.c` https://github.com/Sensirion/embedded-uart-sps/blob/6e3256e8d4e63dcc49a6e5f322467fabfb8a2400/embedded-uart-common/sensirion_shdlc.c#L178

So are there two errors or did you make a typo in one of the messages?

>  `Bytes received: 7E D0 7D 4C BA 43 A6 B2 42 9A 30 4C 13 15 31 2 A6 FF `

When decoding this frame it looks a bit strange. The `0x7D` seems to stuff the `0x4C` byte which would unstuff to `0x6C` which isn't a byte that needs stuffing (only `0x7E`, `0x7D`, `0x11` and `0x13` needs stuffing).
Also the length field would be `0x43` which is too long for the serial number, which has a maximum length of 32. So indeed something seems to go wrong badly.

Do you have access to a logic analyzer to track what is going on on the wire?

hijoumake, 2019-02-22:
> I'm sure that it is failing at L197 
Apologies. This is a typo. I meant to reference the same L178 as previously stated.

>When decoding this frame it looks a bit strange.
>So indeed something seems to go wrong badly.
The output was logged prior to the stuff/unstuff logic being executed, which makes it even stranger. I log it to the Arduino serial console directly after reading bytes from the serial input buffer.

I'll attempt using some of the other digital I/O pins on my development board with the Arduino Software Serial library and report back if that changes anything. If that does not produce significantly different results, I'll likely switch my development effort to the Raspberry Pi hardware that I've been able to get working with the sample code.

> Do you have access to a logic analyzer to track what is going on on the wire?
Unfortunately, I do not have access to a logic analyzer. My best guesses right now are that this strange output is either the result of some unexpected interaction with the Arduino and/or ATmega platform or a hardware issue with my Feather development board hardware serial ports since the sample code seems to run fine (executes without any errors and produces PM concentration readings that appear reasonable given the environment) on my Raspberry Pi Zero W using the same wires/JST connector and SPS30 unit, although obviously in a different software environment.

rnestler, 2019-04-03:
@hijoumake Could you test again with latest master? https://github.com/Sensirion/embedded-uart-sps/pull/32 may be related to your issue and fix your issue.

hijoumake, 2019-04-19:
I've tested the latest master on an Adafruit nRF52840 Express dev board. It seems to be working as expected. I'll close this issue now as I no longer intend to develop on the 32u4 Bluefruit LE .

closed 2019-04-19 by hijoumake