Sensirion/embedded-uart-sps #25: Distingiuish between different SENSIRION_SHDLC_ERR_MISSING_STOP errors

rnestler, 2019-02-20:
We currently return SENSIRION_SHDLC_ERR_MISSING_STOP for three different error cases:
https://github.com/Sensirion/embedded-uart-sps/blob/6e3256e8d4e63dcc49a6e5f322467fabfb8a2400/embedded-uart-common/sensirion_shdlc.c#L180
https://github.com/Sensirion/embedded-uart-sps/blob/6e3256e8d4e63dcc49a6e5f322467fabfb8a2400/embedded-uart-common/sensirion_shdlc.c#L190
https://github.com/Sensirion/embedded-uart-sps/blob/6e3256e8d4e63dcc49a6e5f322467fabfb8a2400/embedded-uart-common/sensirion_shdlc.c#L197
This makes it harder to diagnose errors. We should probably come up with two more error codes to use in these cases.

KaninMcGuire, 2019-11-19:
[image]
Here you can use these and create variables to distinguish the errors

abrauchli, 2019-11-19:
Hi @KaninMcGuire thanks for your input. I think the table you posted is coming from a sensor datasheet like the SPS30 and specify the error codes that are returned from the sensor when the communication protocol is intact - what @rnestler means, is that the SHDLC protocol decoding failed, not the sensor specific commands. You would find that in case of a faulty link (dropped communication, flipped bits, a second device on the bus that doesn't respect SHDLC etc.)

abrauchli, 2019-11-28:
Fixed with https://github.com/Sensirion/embedded-uart-sps/pull/50 (RFC)

rnestler, 2020-03-13:
Fixed with #50.

closed 2020-03-13 by rnestler