Sensirion/embedded-uart-sps #40: Voltage via UART?

djmbritt, 2019-05-19:
So I'm wondering, if using UART with the sps30, does the voltage default to 5v or 3v3?
Looking at the data sheet I see the following:
TTL 5V and LVTTL 3.3V compatible

I assume that when I select to use the I2C protocol it defaults to 3v3 on the pin 2 and 3.
And that when I use the UART protocol it defaults to 5v.

Is this correct?

rnestler, 2019-05-23:
Hi @djmbritt 

> I assume that when I select to use the I2C protocol it defaults to 3v3 on the pin 2 and 3.

I2C needs pull up resistors (Section 5 "Operation and Communication through the I2C Interface" in the datasheet) so it operates at whatever voltage you connect your pull up resistors to.

The inputs are 5V tolerant, but will output 3.3V. So it should work if you connect a device which uses 5V logic levels.

closed 2019-11-15 by abrauchli

Roopesh101, 2020-02-18:
My host is at 1.8V logic. planning to use a level shifter for UART signal from host to SPS30.
If I shift them to 5V, I may not receive a 5V signal from SPS30. I should shift them to 3.3V. In this way I get proper logic levels for my level shifter. Can you confirm on this.

rnestler, 2020-02-19:
> I should shift them to 3.3V. In this way I get proper logic levels for my level shifter. Can you confirm on this.

Yes, shifting to 3.3V should work.