Sensirion/embedded-uart-sps #49: Only getting NaN readings via UART on Raspberry Pi 3

schweini, 2019-11-14:
I hooked up my SPS30 to the Raspberry Pi's GPIO UART pins, and followed the instructions, setting the serial port to /dev/ttyS0 (after disabling serial console in raspi-config).

Running `sps30_example_usage`, the sensor seems to be identified (indicated by the serial number, i think?), but only gives `nan` values, even though the fan seems to spin:

```
# ./sps30_example_usage
SPS30 sensor probing successful
SPS30 Serial: E33C9E11208D72B9
measurements started
measured values:
        nan pm1.0
        nan pm2.5
        nan pm4.0
        nan pm10.0
        nan nc0.5
        nan nc1.0
        nan nc2.5
        nan nc4.5
        nan nc10.0
        nan typical particle size

^C

```

abrauchli, 2019-11-15:
Hi @schweini thanks for the report. On the software side everything seems legit. Have you tried the i2c interface by any chance?

abrauchli, 2019-11-15:
Hi @schweini it seems the issue is on our side and we'll send you a replacement unit. Can you contact me by email andreas.brauchli@sensirion.com ?

closed 2019-11-15 by abrauchli

felipedcaldas, 2021-05-06:
Hello, I have a question about the NaN fault. I am using sensirion temperature and humidity sensors in a historical museum. When downloading data through the My ambiance app, I noticed: 
1) The more readings I take from the same sensor, the less NaN will appear.
2) Depending on the cell phone being used (same version of the app) the frequency of NaN failures also varies.

Could you tell me if there is any way the readings can come without fail? 

Thank you very much for your attention
Felipe

rnestler, 2021-08-03:
Hi @felipedcaldas 

This doesn't seem related to the SPS sensor so this is the wrong place to ask. I recommend you to use the official support channel under https://www.sensirion.com/en/about-us/support-center/