Sensirion/embedded-uart-sps #52: help wanted for SPS30 with UART communication

Mohit2183, 2020-01-11:
we are facing a critical issue with communication with SPS30 WITH PC
we were testing sps30 by connecting with docklight.
we have used usb to serial converter to command the sps30. 
pin 1 was connected to poer supply +5vdc , 
pin 5 was connected to 0vdc. 
pin4 was left open for UART, 
pin 2 was connected to Tx of master (PC), 
pin 3 was connected to Rx of master (pc). 
From docklight v2.3 we had set the com port , 115200, 8N1 communication settings. 
we have placed command in format of Hex codes i.e. 
start command 
Tx: 7E 00 00 01 02 F9 7E , 
Rx REPLY: 7E 00 00 01 02 F9 7E , 
Tx: 7E 00 00 00 FF 7E, 
Rx REPLY: x: 7E 00 00 00 FF 7E , 
Tx: 7E 00 03 00 FC 7E , 
Rx REPLY: 7E 00 03 00 FC 7E , 
Tx: 7E 00 D3 00 2C 7E, 
Rx REPLY: 7E 00 D3 00 2C 7E . 
OVER ALL FAN IS ALSO NOT ROTATING PHYSICALLY.. 
could you pls guide us why sps30 is not giving response to our commands.
how to proceed further with sps30

rnestler, 2020-01-22:
>  Tx: 7E 00 00 01 02 F9 7E ,
This looks wrong, according to the [data sheet](https://www.sensirion.com/fileadmin/user_upload/customers/sensirion/Dokumente/0_Datasheets/Particulate_Matter/Sensirion_PM_Sensors_SPS30_Datasheet.pdf) it should be:
`0x7E 0x00 0x00 0x02 0x01 0x03 0xF9 0x7E`

>  Rx REPLY: 7E 00 00 01 02 F9 7E ,

This indicates the error in the state byte (0x01) which means "Wrong data length for this command (too much or little data)" also according to the data sheet. It is caused by the wrong start measurement command above.

rnestler, 2020-07-02:
I'll close this, since there was no activity. Feel free to re-open if you still have any issues @Mohit2183

closed 2020-07-02 by rnestler