Sensirion/embedded-uart-sps #63: uart-sps

mike197678, 2020-07-06:
Hello,

Sorry I'm a total beginner at raspberry pi.

I saw your code regarding the SPS30 sensor.
I installed your files as best I could

Unfortunately I don't get any values.

if I do "make" I get

`pi@raspberrypi:~/sps30-uart-3.0.0 $ make
rm -f sps30_example_usage
cc -Os -Wall -fstrict-aliasing -Wstrict-aliasing=1 -Wsign-conversion -fPIC -I. -I. -I.  -o sps30_example_usage ./sensirion_arch_config.h ./sensirion_uart.h ./sensirion_shdlc.h ./sensirion_shdlc.c ./sps_git_version.h ./sps_git_version.c ./sps30.h ./sps30.c ./sensirion_uart_implementation.c ./sps30_example_usage.c
pi@raspberrypi:~/sps30-uart-3.0.0 $
`
but not any more.

sorry for my bad English
can you help me?

rnestler, 2020-07-07:
Well this looks like compilation did work, no? Have you tried to continue with the next step and run the executable as in the [guide](https://github.com/Sensirion/embedded-uart-sps/blob/master/docs/getting-started-on-the-raspberry-pi.md#compile-and-run)?

So run `./sps30_example_usage`?

mike197678, 2020-07-07:
oh sorry
I totally forgot and overlooked this command.
`./sps30_example_usage`

Now the sensor is finally working. Perfect

Is it possible to stop the application automatically after e.g. 10 measurements?

Thanks a lot.

Sometimes you don't see the forest in front of trees.

Michael

psachs, 2020-07-08:
You have to modify the sps30_example_usage.c file and replace the `while(1) {..}` block with something like this:
```c
    int16_t i;
    for (i = 0; i < 10; i++) {
        sensirion_sleep_usec(SPS30_MEASUREMENT_DURATION_USEC); /* wait 1s */
        ret = sps30_read_measurement(&m);
        if (ret < 0) {
            printf("error reading measurement\n");

        } else {
            printf("measured values:\n"
                   "\t%0.2f pm1.0\n"
                   "\t%0.2f pm2.5\n"
                   "\t%0.2f pm4.0\n"
                   "\t%0.2f pm10.0\n"
                   "\t%0.2f nc0.5\n"
                   "\t%0.2f nc1.0\n"
                   "\t%0.2f nc2.5\n"
                   "\t%0.2f nc4.5\n"
                   "\t%0.2f nc10.0\n"
                   "\t%0.2f typical particle size\n\n",
                   m.mc_1p0, m.mc_2p5, m.mc_4p0, m.mc_10p0, m.nc_0p5, m.nc_1p0,
                   m.nc_2p5, m.nc_4p0, m.nc_10p0, m.typical_particle_size);
        }
    }
```

rnestler, 2020-07-08:
> Now the sensor is finally working. Perfect

Good to hear :)

> Is it possible to stop the application automatically after e.g. 10 measurements?

Well you could replace the [endless loop](
https://github.com/Sensirion/embedded-uart-sps/blob/4c401385f2a876200a63959e8763e36b039a96c4/sps30-uart/sps30_example_usage.c#L105) in the example with a for loop like @psachs suggested.

Do you intend to let the sensor running in the background? Because if you stop measurements after just 10s, precision won't be that good since the device should be measuring for more than 30s to achieve good results. But to leave it running you'd may need to change the code in the beginning as well to only start measurements if they aren't running already.

> PS: i would like to start and stop the application via node-red and display the output in a debug window.
is that feasible?

I don't have any experience with node-red. But I'm curious how your setup looks like and what you try to achieve :slightly_smiling_face:

mike197678, 2020-07-08:
Hello,

first of all thank you for the great support.

So regarding the measurement, I think that the sensor does not necessarily have to run 24/7.
It's certainly better for life.

It would be enough for me, for example,
one measurement every hour with e.g. 10 or 20 measurements,
because of the accuracy.

or does it not matter that the sensor is running 24/7?

I did that with the code from
Tried @psachs.

Unfortunately:

`pi@raspberrypi:~/release/sps30-uart-3.0.0 $ make
rm -f sps30_example_usage
cc -Os -Wall -fstrict-aliasing -Wstrict-aliasing=1 -Wsign-conversion -fPIC -I. -I. -I.  -o sps30_examp                             le_usage ./sensirion_arch_config.h ./sensirion_uart.h ./sensirion_shdlc.h ./sensirion_shdlc.c ./sps_gi                             t_version.h ./sps_git_version.c ./sps30.h ./sps30.c ./sensirion_uart_implementation.c ./sps30_example_                             usage.c
./sps30_example_usage.c: In function ‘main’:
./sps30_example_usage.c:82:30: error: ‘SPS30_MEASUREMENT_DURATION_USEC’ undeclared (first use in this                              function)
         sensirion_sleep_usec(SPS30_MEASUREMENT_DURATION_USEC); /* wait 1s */
                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
./sps30_example_usage.c:82:30: note: each undeclared identifier is reported only once for each functio                             n it appears in
make: *** [Makefile:10: sps30_example_usage] Error 1
pi@raspberrypi:~/release/sps30-uart-3.0.0 $
`

1000 thanks

my sps30_example_usage.c file:
`#include <stdio.h>  // printf

#include "sensirion_uart.h"
#include "sps30.h"

/**
 * TO USE CONSOLE OUTPUT (PRINTF) AND WAIT (SLEEP) PLEASE ADAPT THEM TO YOUR
 * PLATFORM
 */
//#define printf(...)

int main(void) {
    struct sps30_measurement m;
    char serial[SPS30_MAX_SERIAL_LEN];
    const uint8_t AUTO_CLEAN_DAYS = 4;
    int16_t ret;

    while (sensirion_uart_open() != 0) {
        printf("UART init failed\n");
        sensirion_sleep_usec(1000000); /* sleep for 1s */
    }

    /* Busy loop for initialization, because the main loop does not work without
     * a sensor.
     */
    while (sps30_probe() != 0) {
        printf("SPS30 sensor probing failed\n");
        sensirion_sleep_usec(1000000); /* sleep for 1s */
    }
    printf("SPS30 sensor probing successful\n");

    ret = sps30_get_serial(serial);
    if (ret)
        printf("error %d reading serial\n", ret);
    else
        printf("SPS30 Serial: %s\n", serial);

    ret = sps30_set_fan_auto_cleaning_interval_days(AUTO_CLEAN_DAYS);
    if (ret)
        printf("error %d setting the auto-clean interval\n", ret);

    ret = sps30_start_measurement();
    if (ret < 0)
        printf("error starting measurement\n");
    printf("measurements started\n");

    do {

     int16_t i;
    for (i = 0; i < 10; i++) {
        sensirion_sleep_usec(SPS30_MEASUREMENT_DURATION_USEC); /* wait 1s */
        ret = sps30_read_measurement(&m);
        if (ret < 0) {
            printf("error reading measurement\n");

        } else {
            printf("measured values:\n"
                   "\t%0.2f pm1.0\n"
                   "\t%0.2f pm2.5\n"
                   "\t%0.2f pm4.0\n"
                   "\t%0.2f pm10.0\n"
                   "\t%0.2f nc0.5\n"
                   "\t%0.2f nc1.0\n"
                   "\t%0.2f nc2.5\n"
                   "\t%0.2f nc4.5\n"
                   "\t%0.2f nc10.0\n"
                   "\t%0.2f typical particle size\n\n",
                   m.mc_1p0, m.mc_2p5, m.mc_4p0, m.mc_10p0, m.nc_0p5, m.nc_1p0,
                   m.nc_2p5, m.nc_4p0, m.nc_10p0, m.typical_particle_size);
        }
    }

    } while (1);

    if (sensirion_uart_close() != 0)
        printf("failed to close UART\n");

    return 0;
}
`

abrauchli, 2020-07-08:
Hi @mike197678 

It seems that you're using v3.0.0, could you try with the latest release https://github.com/Sensirion/embedded-uart-sps/releases/latest ?

The sensor is made for continuous operation but there are various reasons to not wanting to do this (e.g. power savings or longevity). However, due to internal calibrations it will yield more precise values when running continuously. Thus, as @rnestler suggested, you should run the sensor for at least 30s (and even more is better) before trusting the values.

mike197678, 2020-07-08:
Hi @abrauchli 

hello, yes exactly, version 3.1.0 works with a minute break. Perfect.
Then I can change the break time.

So now I just have to get the values ​​somehow in node-red.!?

abrauchli, 2020-07-08:
Great, thanks for the reply! I'm closing the issue since we won't be able to support you with node-red.
Feel free to open a new one if you think there's an issue with the driver.

closed 2020-07-08 by abrauchli