Sensirion/embedded-uart-sps #71: Support for Nano 33 BLE Sense

GuntasSinghProg5, 2020-08-01:
Hello all,
I'm trying to use the SPS30 PM sensor with a UART driver from this repository, however, I am new to this sensor and I don't know what to do in order to use the sensor on UART with the Nano 33 BLE Sense, which runs on an nRF52840 MCU. It communicates through the use of Serial1. functions.  What steps do I need to perform in order to use the SPS with UART on this board. I tried using I2C, it works but my project uses much faster pull rates than i2c.
Regards
Guntas

rnestler, 2020-08-06:
Hi @GuntasSinghProg5 

Have you tried the [Arduino sample implementation](https://github.com/Sensirion/embedded-uart-sps/blob/bed826200bf069e993949d81b9b389eeba18cbec/embedded-uart-common/sample-implementations/arduino/sensirion_uart_implementation.cpp)?

Just copy it to the top level folder and maybe change `Serial2` to `Serial1`.

rnestler, 2020-08-06:
> I tried using I2C, it works but my project uses much faster pull rates than i2c.

What do you mean by "faster pull rates"? The device only provides one measurement per second, so you'll get the same values if you just read them again and again.

GuntasSinghProg5, 2020-08-06:
Will try, Mr. Nestler. Thank you for replying.

GuntasSinghProg5, 2020-08-06:
I meant that on Sparkfun, it says that I2C only works fine for mass concentrations and not number, so I think using UART would solve that problem. UART is faster than I2C, I think. So, I want to use the fastest protocol possible for my project.

Regards
Guntas

closed 2020-10-31 by GuntasSinghProg5