Sensirion/embedded-uart-sps #75: 40: internal function argument out of range

jaapliefting, 2020-11-10:
For new development of telemetric particle counter we are looking into SPS30 and made some test software on PC to do the uart SHDLC protocol to communicate with the SPS30

For all commands sent to it, sps30 returns 40: internal function argument out of range status response. However, the command seems to be executed by the sensor.  The fan actually switches on after start command, and off after stop command.

Is this an issue with the protocol or is it something internal in de sensor ?

jaapliefting, 2020-11-10:
A true power-down power-up sequence of the sensor seems to have solved the issue.

closed 2020-11-10 by jaapliefting

abrauchli, 2020-11-11:
Hi @jaapliefting 

Thanks for reporting, we'll check internally if we have something on the record for this behavior. I'm just curious if you have also tried the soft reset (`sps30_reset`)? I see that the documentation for the i2c driver (https://github.com/Sensirion/embedded-sps/blob/d965ff6b6c79d61915dafb5e657f79726bd626c3/sps30-i2c/sps30.h#L230) is much better than the UART one..