Sensirion/embedded-uart-sps #77: Output format is float values only — integer format not supported

rwaldura, 2020-12-05:
Datasheet says an output format must be specified when starting the measurement with the “Start Measurement command”; 2 output formats are supported:

> Measurement Output Format:
> 0x03: Big-endian IEEE754 float values
> 0x05: Big-endian unsigned 16-bit integer values

It seems this implementation supports only the float format today. 

https://github.com/Sensirion/embedded-uart-sps/blob/master/sps30-uart/sps30.c#L40
#define SPS30_SUBCMD_MEASUREMENT_START  { 0x01, 0x03 }

What is the benefit of receiving float values from the sensor? Datasheet says sensor precision is never better than 10 μg/m3. I.e. fractional digits in float values could be considered "unwanted" precision, and may not mean much. 

Am I correct to understand the default float output format is in fact not more precise than integer format? And that, for all intents and purposes, float values should be rounded to integers?   

"Datasheet SPS30 - Particulate Matter Sensor for Air Quality Monitoring and Control"

Mass concentration precision1,2 for PM1 and PM2.53 0 to 100 μg/m3 ±10 μg/m3
100 to 1000 μg/m3
Mass concentration precision1,2 for PM4, PM104 0 to 100 μg/m3 ±25 μg/m3
100 to 1000 μg/m3

abrauchli, 2020-12-05:
Hi @rwaldura

You're absolutely correct in your assessment. As far as I'm aware, the float output was simply there first during development and we stuck with it.
The choice of output format was provided, IIRC, because some platforms have buffer limits (e.g. we noticed some Arduinos). Using a `uint16_t`, we're thus shrinking the demands on the buffer size. Otherwise, floats might be easier if that's the type you intend to convert to anyway for your application.

I'll close the issue, but you're welcome to open it again if you have further questions.

Kind regards
Andreas

closed 2020-12-05 by abrauchli

rwaldura, 2020-12-06:
Thank you for this clear explanation, much appreciated.