Sensirion/embedded-uart-sps #85: SPS30 with STM32F401

Yousaf93, 2023-12-19:
Hi, 
I have ported the code as per guidlines but as far my Knowledge Simple HALL_UART_RECEIVE would not work as the RX data size is not accurately known for each command and also the given library sends max data len to HAL , So i used HAL_UART_READtoIDLE which is solely for the given purpose but still get timeout error and does not receive any data 

> int16_t sensirion_uart_rx(uint16_t max_data_len, uint8_t* data) {
>     // TODO: implement
> 	uint16_t actual_bytes=0;
> 	HAL_StatusTypeDef status;
> 	status = HAL_UARTEx_ReceiveToIdle(&huart2, data,max_data_len, &actual_bytes,1000);
> 	switch(status) {
> 	        case HAL_OK:
> 	            printf("Status: HAL_OK\n");
> 	            break;
> 	        case HAL_ERROR:
> 	            printf("Status: HAL_ERROR\n");
> 	            break;
> 	        case HAL_BUSY:
> 	            printf("Status: HAL_BUSY\n");
> 	            break;
> 	        case HAL_TIMEOUT:
> 	            printf("Status: HAL_TIMEOUT\n");
> 	            break;
> 	        default:
> 	            printf("Status: UNKNOWN\n");
> 	            break;
> 	    }
> 	//if(HAL_UART_Receive(&huart2, &data,max_data_len,HAL_MAX_DELAY)== HAL_OK)
> 	if(status== HAL_OK)
> 	{
> 		return (int16_t)actual_bytes;
> 	}
> 	else
> 	{
> 		return -1;
> 	}
> }

abrauchli, 2023-12-19:
Hi @Yousaf93, 

Great to see an implementation. I quickly glanced over it and you're right that the timeout variant is better in this case, however I do have a few suggestions:
* running into a timeout could (and typically will) mean that the read was successful (with less than `max_data_len` bytes), so I'd suggest always returning the actual number of bytes if there's been a write to `data`.
* Make the timeout a `#define` - one second seems rather high given that we'll likely run into it most of the reads.
* I'd wrap the `printf(...)` in a #ifdef DEBUG to avoid interfering with user output from the users of the library but I'll let the reviewers take their stance on that.
* Lastly, it'd greatly help the reviewers if you could you resubmit the whole implementation (not just the rx function) in a folder in https://github.com/Sensirion/embedded-uart-sps/tree/master/embedded-uart-common/sample-implementations in a new pull request rather than copying it into this issue and apply proper indentation and formatting (I think the project has a .clang-format you can use to apply the project style automatically - there are quite a few whitespace errors which make the code hard to read even when indented). It'd also help if you wrote how exactly you tested your implementation (board you mentioned, sensor (SPS30?), connection setup to the board (if there's more than one UART port), what worked, what didn't and what wasn't tested).

Merry Christmas,
Andy

Yousaf93, 2023-12-20:
Hi @abrauchli , Thank you for your guidance 
I would have definitely opted to the method mentioned but unfortunately i have strict timeline regarding the work so i did the work directly. But i would share the outcomes of the current implementations and would try to share the implementations later.
The current functions, Consider the Simple UART configurations and Poll for Serial data after transmit , 
As the SPS30 works on Master Slave MOSI MISO technique it responds as soon receives a command and with polling the reponse is missed which yields to timout error
So modificaction in following function is needed in place of HAL_UART_RX , The lenght data should be passed form the ISR of UART which needs to be initilaized in main 
int16_t sensirion_shdlc_rx(uint8_t max_data_len,
                           struct sensirion_shdlc_rx_header* rxh,
                           uint8_t* data) {
    int16_t len;
    uint16_t i;
    uint8_t rx_frame[SHDLC_FRAME_MAX_RX_FRAME_SIZE];
    uint8_t* rx_header = (uint8_t*)rxh;
    uint8_t j;
    uint8_t crc;
    uint8_t unstuff_next;

    len = sensirion_uart_rx(2 + (5 + (uint16_t)max_data_len) * 2, rx_frame);
    //printf("Total %d bytes received \n", len);

    if (len < 1 || rx_frame[0] != SHDLC_START)
            return SENSIRION_SHDLC_ERR_MISSING_START;

        for (unstuff_next = 0, i = 1, j = 0; j < sizeof(*rxh) && i < len - 2; ++i) {
            if (unstuff_next) {
                rx_header[j++] = sensirion_shdlc_unstuff_byte(rx_frame[i]);
                unstuff_next = 0;
            } else {
                unstuff_next = sensirion_shdlc_check_unstuff(rx_frame[i]);
                if (!unstuff_next)
                    rx_header[j++] = rx_frame[i];
            }
        }
        if (j != sizeof(*rxh) || unstuff_next)
            return SENSIRION_SHDLC_ERR_ENCODING_ERROR;

        if (max_data_len < rxh->data_len)
            return SENSIRION_SHDLC_ERR_FRAME_TOO_LONG; /* more data than expected */

        for (unstuff_next = 0, j = 0; j < rxh->data_len && i < len - 2; ++i) {
            if (unstuff_next) {
                data[j++] = sensirion_shdlc_unstuff_byte(rx_frame[i]);
                unstuff_next = 0;
            } else {
                unstuff_next = sensirion_shdlc_check_unstuff(rx_frame[i]);
                if (!unstuff_next)
                    data[j++] = rx_frame[i];
            }
        }

        if (unstuff_next)
            return SENSIRION_SHDLC_ERR_ENCODING_ERROR;

        if (j < rxh->data_len)
            return SENSIRION_SHDLC_ERR_ENCODING_ERROR;

        crc = rx_frame[i++];
        if (sensirion_shdlc_check_unstuff(crc))
            crc = sensirion_shdlc_unstuff_byte(rx_frame[i++]);

        if (sensirion_shdlc_crc(rxh->addr + rxh->cmd + rxh->state, rxh->data_len,
                                data) != crc)
            return SENSIRION_SHDLC_ERR_CRC_MISMATCH;

        if (i >= len || rx_frame[i] != SHDLC_STOP)
            return SENSIRION_SHDLC_ERR_MISSING_STOP;

        return 0;
}

LeonieFierz, 2024-03-13:
@Yousaf93 thank you for sharing the code. If you find time to submit the whole implementation as mentioned by abrauchli please open a new pull request.

closed 2024-03-13 by LeonieFierz